tu-darmstadt-ros-pkg/hector_quadrotor #37: xacro:macro tag missing in quadrotor_sensors.gazebo.xacro. Causes launch issues.

krrish94, 2015-07-06:
I'm using the hector_quadrotor stack with ROS Jade, but I'm quite sure this issue persists with other versions as well. When I try launching spawn_quadrotor.launch (from the hector_quadrotor_gazebo package) using `$ roslaunch hector_quadrotor_gazebo spawn_quadrotor.launch`, it fails with the following error messages.

```
Traceback (most recent call last):
  File "/opt/ros/jade/share/xacro/xacro.py", line 60, in <module>
    xacro.main()
  File "/opt/ros/jade/lib/python2.7/dist-packages/xacro/__init__.py", line 693, in main
    eval_self_contained(doc, in_order)
  File "/opt/ros/jade/lib/python2.7/dist-packages/xacro/__init__.py", line 616, in eval_self_contained
    eval_all(doc.documentElement, macros, symbols)
  File "/opt/ros/jade/lib/python2.7/dist-packages/xacro/__init__.py", line 584, in eval_all
    raise XacroException("unknown macro name: %s" % node.tagName)
xacro.XacroException: unknown macro name: xacro:quadrotor_sensors
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/jade/share/xacro/xacro.py '/home/turing/catkin_ws/src/hector_quadrotor_description/urdf/quadrotor.gazebo.xacro' base_link_frame:=/base_link world_frame:=world] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(arg model)' base_link_frame:=$(arg base_link_frame) world_frame:=$(arg world_frame)" name="robot_description"/>
The traceback for the exception was written to the log file

```

Clearly, there was something wrong with hector_quadrotor_description/urdf/quadrotor_plugins.gazebo.xacro. I dug deeper and found that there was a missing 
```
<xacro:macro name="quadrotor_sensors">
....
</xacro:macro>
```
pair of tags in the file hector_quadrotor_description/urdf/quadrotor_sensors.gazebo.xacro.

After adding the tags, it launched fine. I'm starting this issue so that someone makes the corresponding change to the repository.

closed 2015-11-07 by meyerj

krrish94, 2015-11-07:
Thanks for the fix!

Regards
Krishna

meyerj, 2015-11-07:
Thanks for the patch!