tu-darmstadt-ros-pkg/hector_quadrotor #45: C++11 Required for catkin_make

kuehnma, 2015-12-04:
Hey!

following the outdoor flight tutorial for hector_quadrotor I might ran into an issure with different C++ compiler versions. Checking out all repos with the rosinstall file from https://raw.github.com/tu-darmstadt-ros-pkg/hector_quadrotor/indigo-devel/tutorials.rosinstall works fine.

On Ubuntu 14.04 with ROS indigo and Gazebo6.5, running catkin_make fails as my default compiler version is not C++11. After adding ```set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")``` in the CMakeLists.txt of
hector_gazebo_plugins
hector_gazebo_thermal_camera
hector_quadrotor_controller_gazebo
hector_quadrotor_gazebo_plugins 
it works fine. Not sure if this is the best solution.

Best regards

Markus

meyerj, 2015-12-09:
This issue has been solved in:
* hector_quadrotor/indigo-devel: https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/commit/474a9f3d1090b374de3b012a3a20afbb801c5eb8
* hector_quadrotor/jade-devel: https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/commit/340b29702d8ac135f408713be2565e9884790800
* hector_gazebo/indigo-devel: https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/commit/22e73b3beecbe8f3c14b2cd0162bc1cf7f4d5c65 (see also https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/pull/22)
* hector_gazebo/jade-devel: https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/commit/bdeea2542b0fb53ab0022d4e0557ef892f840afa

Thanks for reporting!

closed 2015-12-09 by meyerj

kuehnma, 2015-12-09:
Great! Thanks for fixing it!