tu-darmstadt-ros-pkg/hector_quadrotor #46: Spining Holuyo Lidar

fkendoul, 2015-12-17:
I am working on a project that involves a spinning Hokuyo lidar. Do the current packages offer the option to spin the Lidar? 
 - If NO, what's the most efficient/easiest way to do that (i.e., what files to modify exactly?).

Thank you.

libing64, 2016-03-08:
I tried to add the spinning Hokuyo lidar to the quadrotor, 

[spawn_quadrotor_with_spinning_lidar.urdf.xacro.txt](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/files/162537/spawn_quadrotor_with_spinning_lidar.urdf.xacro.txt)

However, the transformation between ${name}_spinning_lidar_spin_link and ${name}_spinning_lidar_root_link is lost.
[image]
[image]
[image]

It seems that the transformation is defined in spinning_hokuyo_utm30lx.urdf.xacro, but why it doesn't work?

libing64, 2016-03-08:
In gazebo, the lidar is on the quadrotor and rotating, however,  in rviz, the lidar is not on the quadrotor because the transformation between xx_spinning_lidar_root_link and xx_spinning_lidar_root link is lost.

libing64, 2016-03-13:
I have solved this problem in #53, and the urdf and lanuch file are in my forked repo.