tu-darmstadt-ros-pkg/hector_quadrotor #47: rviz error in hector_quadrotor_demo

AzharudeenMohammad, 2016-01-13:
I have source installed hector quadrotor demo .When i roslaunch outdoor_ flight_gazebo the gazebo window pops up with the simulation but the rviz gets killed..i cant figure out why this happens ...
the error o/p is as follows:

[rviz-8] process has died [pid 29273, exit code -11, cmd /home/azar/ros_catkin_ws/install_isolated/lib/rviz/rviz -d /home/azar/hector_quadrotor_tutorial/src/hector_quadrotor/hector_quadrotor_demo/rviz_cfg/outdoor_flight.rviz __name:=rviz __log:=/home/azar/.ros/log/b02e4b10-ba10-11e5-9ade-c038966edbf9/rviz-8.log].
log file: /home/azar/.ros/log/b02e4b10-ba10-11e5-9ade-c038966edbf9/rviz-8*.log

nicolaerosia, 2016-02-02:
What's in the log file?

AzharudeenMohammad, 2016-03-16:
This problem was solved later when I reinstalled ubuntu+ros

closed 2016-03-16 by AzharudeenMohammad