tu-darmstadt-ros-pkg/hector_quadrotor #50: Could not find a package configuration file provided by   "hector_pose_estimation" with any of the following names

MetaMemoryT, 2016-02-18:
```
-- +++ processing catkin package: 'hector_quadrotor_pose_estimation'
-- ==> add_subdirectory(hector_quadrotor/hector_quadrotor_pose_estimation)
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "hector_pose_estimation" with any of the following names:

    hector_pose_estimationConfig.cmake
    hector_pose_estimation-config.cmake

  Add the installation prefix of "hector_pose_estimation" to
  CMAKE_PREFIX_PATH or set "hector_pose_estimation_DIR" to a directory
  containing one of the above files.  If "hector_pose_estimation" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  hector_quadrotor/hector_quadrotor_pose_estimation/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/sean/p/hector/build/CMakeFiles/CMakeOutput.log".
See also "/home/sean/p/hector/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

Leslie-Fang, 2016-03-07:
same problem when catkin_make in jade! have you solved it?

libing64, 2016-03-07:
hector_pose_estimation package is in the following project,
https://github.com/tu-darmstadt-ros-pkg/hector_localization
compile hector_pose_estimation first and then compile hector_quadrotor

Leslie-Fang, 2016-03-07:
Sorry,I still got the same problem. Here is what I have done

Firstly I have catkin_make the package of 
             https://github.com/tu-darmstadt-ros-pkg/hector_localization  
in the master branch.

Then I catkin_make the package 
    https://github.com/Leslie-Fang/hector_quadrotor.git
in the jade-devel package~~~

Is there any tutorial about how to use hector related package?

meyerj, 2016-03-07:
See http://wiki.ros.org/hector_quadrotor/Tutorials.

> Firstly I have catkin_make the package of 
> https://github.com/tu-darmstadt-ros-pkg/hector_localization
> in the master branch.

You need branch catkin of [hector_localization](https://github.com/tu-darmstadt-ros-pkg/hector_localization) (or install dependencies as binary packages).

There is a rosinstall file called hector_quadrotor.rosinstall in each of the hector_quadrotor branches with all the dependencies:
* ROS indigo: https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/indigo-devel/hector_quadrotor.rosinstall
* ROS jade: https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/jade-devel/hector_quadrotor.rosinstall

The variant [tutorials.rosinstall](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/indigo-devel/tutorials.rosinstall) mentioned in the ROS wiki additionally contains hector_slam and is only required to launch the demos in package hector_quadrotor_demo.

In general, for all catkin-based ROS source builds, [rosdep](http://wiki.ros.org/rosdep) is the tool of choice to make sure that all dependencies are installed:
```
rosdep install --from-paths src --ignore-src
```
This command would install the released binary packages of all dependencies if invoked in a workspace with only hector_quadrotor. Unfortunately this variant of the `rosdep install` command is not very well documented, although it is the most commonly used since catkin has been introduced, since `rosdep install <package name>` would only work **after** the workspace has been successfully built and the generated `setup.sh` has been sourced.

byronxubin, 2017-02-19:
Hi, 

I've tried what you said, which is to use `rosdep`. While when I want to use this command, there is also an error. Could you help me find out what is wrong here?
[image]

meyerj, 2017-02-20:
Which Ubuntu distro are you using? ROS kinetic is only available for Ubuntu Wily and Xenial. If you have an older Ubuntu version installed, you should use ROS indigo or jade.

Or you simply have to invoke `sudo apt-get update` once before calling rosdep. `ros-kinetic-gazebo-ros-control` should be available:
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=gazebo_ros_control

byronxubin, 2017-02-20:
@meyerj  Hi,

I am currently using Xenial. 

I have tried your solution, it still came out the same result, that it cannot locate that package. Is there other way to install?

nm46nm, 2017-11-19:
thanks, @meyerj  the command  
` rosdep install --from-paths src --ignore-src`
is very useful to solve the dependencies problem.

suraj2596, 2018-11-10:
For lunar users : `rosdep install --from-path src --ignore-src --rosdistro lunar`