tu-darmstadt-ros-pkg/hector_quadrotor #59: ros kinetic compile error?gazebo version not supported?

chesternimiz, 2016-10-16:
**I'm working with ros kinetic. When I compile the hector_quadrotor_kinetic_devel, I got errors like**<br/>
hector_quadrotor-kinetic-devel/hector_gazebo_plugins/src/servo_plugin.cpp:203:25: error: ‘class gazebo::physics::Joint’ has no member named ‘SetMaxForce’
     servo[FIRST].joint->SetMaxForce(0, maximumTorque);
I guess that ros kinetic using gazebo 7 causes some problem. How to fix it? Or is it possible to install a binary version using apt-get install  ros-kinetic-hector-quadrotor* ? Looking forward to your help.

chesternimiz, 2016-10-18:
**I fixed gazebo problems by changing SetMaxForce(a,b) to SetParam("fmax",a,b).
However, the compilation comes into another problem with Eigen. Errors are like**

/usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:723:40: error: ‘Flags’ is not a member of ‘Eigen::internal::evaluator<const hector_pose_estimation::ColumnVector_<-1> >’
     Flags0 = evaluator<ArgType>::Flags & ( (HereditaryBits & ~RowMajorBit) |

Is it because of the version change of Eigen or something else?

meyerj, 2016-11-01:
Please check that you have cloned the latest kinetic-devel branches of hector_quadrotor and all its dependencies (check [hector_quadrotor.rosinstall](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/kinetic-devel/hector_quadrotor.rosinstall)). The released version 0.5.0 of hector_gazebo_plugins should also work and builds successfully in kinetic with Gazebo 7.

The SetMaxForce bug has been solved in https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/pull/24 and merged in June before the release into ROS kinetic.

chesternimiz, 2016-11-03:
Thanks a lot! I've compiled successfully with your help. I launched a world by roslaunch hector_quadrotor_demo outdoor_flight_gazebo.launch. Gazebo started and the quadrotor was spawned. However, when I published messages to cmd_vel, the quadrotor didn't move at all. 

There were many warnings after outdoor_flight_gazebo.launch had been launched.
[ WARN] [1478161709.029377631, 1.000000000]: No command received for 1s, triggering estop
[ WARN] [1478161710.178804839, 2.000000000]: No command received for 2s, triggering estop
[ WARN] [1478161711.282482270, 3.000000000]: No command received for 3s, triggering estop
What's wrong with it?

braraki, 2016-11-05:
I have this same problem on kinetic with Gazebo 7. The quadrotor will not respond to messages on cmd_vel.

byronxubin, 2017-02-20:
Hi @chesternimiz, 

I am new to ROS and I am trying to install hector_quadrotor on Kinetic, when I use catkin_make to compile the package, errors keep up coming out. I really don't know how to do it. I have git cloned the whole package and use wstool to install the other dependencies, then I use catkin_make. Is this right? Could you help explain how to install it?

hvpandya, 2017-03-07:
@byronxubin [ROS ANSWERS](http://answers.ros.org/questions/) is the right place to get help with installation and getting started.

@braraki @meyerj I can confirm that kinetic_devel with Gazebo 7 is still having issues with accepting cmd_vel.

DThought, 2017-03-13:
@hvpandya I am running into the same issue. Has anyone found a workaround for running Hector on kinetic?

meyerj, 2017-03-13:
Kinetic release is still pending. For all those who report that the quadrotor does not accept velocity commands, check https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/issues/66#issuecomment-278938410.

I think the original issue was solved by updating dependencies, so I am closing this one.

closed 2017-03-13 by meyerj