tu-darmstadt-ros-pkg/hector_quadrotor #60: ResourceNotFound: hector_quadrotor_gazebo

braraki, 2016-11-05:
I'm running kinetic on Ubuntu 16.04 with Gazebo 7. I used apt to install the ros-kinetic-hector binaries, but when I run either of the launch files for the indoor or outdoor slam demos, I get the error ResourceNotFound: hector_quadrotor_gazebo. When I look in the /opt/ros/kinetic/share folder, I see that I have installed packages with names such as hector_gazebo and hector_components_description, but not the packages specified in the launch file such as hector_quadrotor_gazebo or hector_quadrotor_description. Am I missing something?

byronxubin, 2017-02-20:
@braraki Hi,

Could you tell me how to use apt-get to install hector_quadrotor on kinetic? Mine seems not working. Thanks in advance.

Akashleena, 2021-11-01:
sudo apt-get install synpatic

Open synaptic package manager and type hector. Install all ros-kinetic-hector packages [
[image]
](url)