tu-darmstadt-ros-pkg/hector_quadrotor #65: Can't obtain a 3D image from DepthImage (compressedDepth issue).

Alvarodamat, 2017-02-02:
Greetings, dear developers. First, thanks for this useful package, I haven't seen a better quadrotor sim before!
Now, the issue: Recently I got your kinetic-devel branch of hector_quadrotor (downloaded from source) in order to obtain 3D images for an assignment. I got it, I got all the dependencies and stuff and ran the indoor SLAM demo, and, when I add a Depth Image (compressedDepth topic) I get this message in the terminal:

Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).

I'm quite a newbie in this, so I don't know if it is a bug or a problem of mine, and of course, I don't know if there is any solution.  

My environment is Ubuntu 16.04 Xenial, I'm ussing ROS Kinetic Kame version, Gazebo 7.0, and RViz 1.12 for Kinetic. 

Some screenshots:

[image]

[image]

[image]

meyerj, 2017-02-02:
I think the reason is that the default variant spawned from `indoor_slam_gazebo.launch` only has a standard RGB camera installed which uses the default `gazebo_ros_camera` plugin from https://github.com/ros-simulation/gazebo_ros_pkgs and does not produce depth images. The demo uses a 2D laser range finder only.

In order to spawn a quadrotor with a depth camera, you should look at the various launch file and robot description options in [hector_quadrotor_gazebo](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/tree/indigo-devel/hector_quadrotor_gazebo/launch) or create your own drone model using [urdf](http://wiki.ros.org/urdf) and appropriate [Gazebo plugins](http://gazebosim.org/tutorials?tut=ros_gzplugins). Probably [spawn_quadrotor_with_kinect.launch](https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/indigo-devel/hector_quadrotor_gazebo/launch/spawn_quadrotor_with_kinect.launch) is more or less what you want (without all the SLAM overhead):

```sh
roslaunch hector_gazebo_worlds willow_garage.launch &
roslaunch hector_quadrotor_gazebo spawn_quadrotor_with_kinect.launch &
```

Typically you would assemble your own application-specific launch file from the examples in the repository or other sources.

Not sure why [rviz](http://wiki.ros.org/rviz) offers a selection of compression methods for depth images for the `/camera/image` topic, but the [image_transport](http://wiki.ros.org/image_transport) interface is quite generic and can contain different types of images, so the subscriber cannot know. The error message clearly states that the published format is `rgb8`, so no depth information.

Please consider to use http://answers.gazebosim.org/ for future questions which are not bug reports or pull requests.

Alvarodamat, 2017-02-03:
Oh, thank you very much! Now I got the model running. And sorry about this thread here, if I need help again, I will head to Answers Gazebo. Thanks again!

closed 2017-02-03 by Alvarodamat