tu-darmstadt-ros-pkg/hector_quadrotor #66: Quadrotor not responding to cmd_vel (with ROS Kinetic and Gazebo 7)

DianeBury, 2017-02-10:
Working with ROS Kinetic and Gazebo 7.

I downloaded the source from the kinetic-devel branch. It compiles fine, but when I launch the demo, I can't make the quadrotor move. I can see the commands being published on cmd_vel. I also checked that gazebo subscribes to cmd_vel and command/twist. Problem doesn't come from the joystick either.
It seems that the callback of the velocity controller is being called but there's no modification of the drone position.

At first I thought the problem was that I needed Matlab to compile, but I get the same result with or without.

Other idea is I think the action takeoff has to be called, but I don't know how, it doesn't seem to be linked to anything.

meyerj, 2017-02-10:
This problem already has been reported in #62. The cause is that, starting from kinetic-devel, the quadrotor spawns with motors disabled. You have to call the `enable_motors` service with the argument set to true, e.g.
```sh
# namespace is optional and depends on how you spawned the simulated quadrotor
rosservice call [namespace/]enable_motors true
```
This "feature" was introduced in https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/pull/57, but probably we should revert the default behaviour to spawn with motors enabled for backwards compatibility.

The action servers provided by `hector_quadrotor_actions` indeed already enable/shutdown motors internally, so they would be an alternative. With the teleop package `hector_quadrotor_teleop` the functions to enable/shutdown the motors can be assigned to a button depending on the mode.

DianeBury, 2017-02-10:
Thanks, works fine now! It makes sense to use this service but the information was hard to find, would be great to make it more obvious for other people.

closed 2017-02-10 by DianeBury

nm46nm, 2017-11-22:
When I run 
rosservice call enable_motors true
there is error like this:
ERROR: Unable to load type [hector_uav_msgs/EnableMotors].
Have you typed 'make' in [hector_uav_msgs]?
What should I do?

YCH188, 2018-03-03:
I tried this method.And found a problem.rosservice call enable_motors true
ERROR: Unable to load type [hector_uav_msgs/EnableMotors].
Have you typed 'make' in [hector_uav_msgs]?
How can I fix it

YCH188, 2018-03-03:
@DianeBury @meyerj Now the problem change to ERROR: Service [/enabel_motors] is not available.

YCH188, 2018-03-03:
@nm46nm Thanks for your help, this is the reason.

yeknafar, 2019-05-09:
@meyerj 
when I tried to enable motor, I got this erro:
"ERROR: Service [/enable_motors] is not available."

I aslo source devel/setup.bash

is there another way to enable the motor ?
Thank you

lcucos, 2019-10-31:
I  am using Ros Melodic with Gazebo 9 and I am having the same problem: 

"Service [/enable_motors]" never becomes available and furthermore I can't enable the motors

I sourced devel/setup.bash. How do I troubleshoot from here?

Thank you

[ INFO] [1572522752.042774160]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1572522752.044541737]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1572522752.393813, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1572522752.412064, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1572522753.020509, 0.015000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1572522753.033414114, 0.027000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1572522753.071783856, 0.064000000]: waitForService: Service [/enable_motors] has not been advertised, waiting...
[ INFO] [1572522753.075779768, 0.068000000]: waitForService: Service [/enable_motors] has not been advertised, waiting...
[ INFO] [1572522753.578911146, 0.195000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1572522753.582273124, 0.195000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1572522753.678428547, 0.195000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1572522753.678517815, 0.195000000]: Starting Laser Plugin (ns = /)
[INFO] [1572522753.679642, 0.195000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1572522753.679763045, 0.195000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1572522753.700172417, 0.207000000]: Physics dynamic reconfigure ready.
[spawn_robot-10] process has finished cleanly
log file: /home/user/.ros/log/1cd5298e-fbd3-11e9-980e-3ff1d43d58a0/spawn_robot-10*.log
[WARN] [1572522781.970296, 21.492000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_spawner-5] process has finished cleanly
log file: /home/user/.ros/log/1cd5298e-fbd3-11e9-980e-3ff1d43d58a0/controller_spawner-5*.log
[rviz-14] process has finished cleanly
log file: /home/user/.ros/log/1cd5298e-fbd3-11e9-980e-3ff1d43d58a0/rviz-14*.log
>>>>>>>>>> gets stuck here <<<<<<<<<<<<