tu-darmstadt-ros-pkg/hector_quadrotor #69: unable to locate ros-hydro-hector-quadrotor-demo packages

sigmorew, 2017-03-18:
Every time I run 
`sudo apt-get install ros-hydro-hector-quadrotor-demo`
I get 
`E: Unable to locate package ros-hydro-hector-quadrotor-demo`

pattrickrice, 2017-12-21:
Experiencing the same problem

shlapion, 2018-01-10:
Hi @sigmorew  and @pattrickrice  depending on which distribution of ROS you have installed on your system, you need to change the name accordingly. 
For instance for Kinetic it should be: 

sudo apt-get install ros-kinetic-hector-quadrotor-demo 

However the package is not available for ROS Kinetic. You should try to install form source.

thecountoftuscany, 2018-09-25:
Hi @shlapion I am experiencing the same problem for `ros-kinetic-hector-quadrotor-demo` on Ubuntu 16.04 (I have **ROS Kinetic** installed)

Terminal output:
```
> sudo apt install ros-kinetic-hector-quadrotor-demo 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-hector-quadrotor-demo
```