tu-darmstadt-ros-pkg/hector_quadrotor #73: CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):

DC-12, 2017-05-31:
**While running the following commands:** 
mkdir ~/hector_quadrotor_tutorial
cd ~/hector_quadrotor_tutorial
wstool init src https://raw.github.com/tu-darmstadt-ros-pkg/hector_quadrotor/kinetic-devel/tutorials.rosinstall
catkin_make
source devel/setup.bash

**I got the following error:** 

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  hector_localization/hector_pose_estimation_core/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'geographic_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  hector_localization/hector_pose_estimation_core/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dc_12/hector_quadrotor_tutorial/build/CMakeFiles/CMakeOutput.log".
See also "/home/dc_12/hector_quadrotor_tutorial/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

raywang-nx, 2017-08-04:
sudo apt-get install ros-kinetic-geographic-msgs

krish007k, 2019-05-25:
how did u solve this error?
I am also getting the same error. can u help me on this? 
Thanq

suqianxin, 2019-08-22:
I got the similar problem as "
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by "controller_interface"
with any of the following names:

controller_interfaceConfig.cmake
controller_interfaceconfig.cmake"

Also,i try ' sudo apt-get install ros-kinetic-controller-interface ' but it did't work.
The answer just as "
The following packages have unmet dependencies:
ros-kinetic-controller-interface : Depends: ros-kinetic-hardware-interfac but it is not going to be installed.
"
What should i do?This problem bother me several days.

robotUR10, 2019-08-30:
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "rospp" with any of
  the following names:

    rosppConfig.cmake
    rospp-config.cmake

  Add the installation prefix of "rospp" to CMAKE_PREFIX_PATH or set
  "rospp_DIR" to a directory containing one of the above files.  If "rospp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  learning_communication/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/book/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/book/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Dear, Sir/Madam
Who has solved this error question? Thanks for your help.

BridgeQZH, 2019-10-08:
I guess if you meet a error message with find_package. You can always choose to install the package you are missing. For example,
If the message shows:
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
Could not find a package configuration file provided by "geographic_msgs"
with any of the following names:

geographic_msgsConfig.cmake
geographic_msgs-config.cmake

Then you should type

`sudo apt-get install ros-kinetic(or melodic)-geographic-msgs`

(It depends on the version of ROS you choose)

If the error message goes in this way:
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by "rospp" with any of
the following names:

rosppConfig.cmake
rospp-config.cmake

Try to type this:

`sudo apt-get install ros-melodic-roscpp`

Just the package name before the Config.cmake

elaiyarajathaangaraju, 2019-11-08:
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "moveit_kinematics"
  with any of the following names:

    moveit_kinematicsConfig.cmake
    moveit_kinematics-config.cmake

Any one have idea ? how to solve this issue. Thanks

StefanFabian, 2019-11-08:
@elaiyarajathaangaraju Maybe read the comment of the guy who posted before you?

```
sudo apt install ros-melodic-moveit-kinematics
```
----------------
Closing and locking this conversation since the original issue was solved and if what @BridgeQZH  wrote does not solve your issue, you should create a new issue instead of commenting here anyway.

closed 2019-11-08 by StefanFabian