tu-darmstadt-ros-pkg/hector_quadrotor #78: catkin_make

yueshukun, 2017-08-09:
Base path: /home/yueshukun/hector_quadrotor_tutorial
Source space: /home/yueshukun/hector_quadrotor_tutorial/src
Build space: /home/yueshukun/hector_quadrotor_tutorial/build
Devel space: /home/yueshukun/hector_quadrotor_tutorial/devel
Install space: /home/yueshukun/hector_quadrotor_tutorial/install
####
#### Running command: "make cmake_check_build_system" in "/home/yueshukun/hector_quadrotor_tutorial/build"
####
####
#### Running command: "make -j8 -l8" in "/home/yueshukun/hector_quadrotor_tutorial/build"
####
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawRC
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Altimeter
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_VelocityXYCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ThrustCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorStatus
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawMagnetic
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_HeightCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_HeadingCommand
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Compass
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ServoCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawImu
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Supply
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_PositionXYCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ControllerState
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_AttitudeCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RC
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorPWM
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RuddersCommand
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_VelocityZCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_YawrateCommand
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetSearchPosition
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetDistanceToObstacle
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetRobotTrajectory
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetRecoveryInfo
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetNormal
[  0%] Built target _catkin_empty_exported_target
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[... 1304 lines cut ...]
make[1]: *** [hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_reset_plugin.dir/all] Error 2
[  8%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/filter/ekf.cpp.o
[  8%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/system.cpp.o
[  8%] Linking CXX executable /home/yueshukun/hector_quadrotor_tutorial/devel/lib/hector_imu_tools/pose_and_orientation_to_imu_node
[  8%] Built target pose_and_orientation_to_imu_node
[  8%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/system/generic_quaternion_system_model.cpp.o
[  9%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/system/ground_vehicle_model.cpp.o
[  9%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/system/imu_model.cpp.o
[  9%] Linking CXX executable /home/yueshukun/hector_quadrotor_tutorial/devel/lib/message_to_tf/message_to_tf
[  9%] Built target message_to_tf
[  9%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurement.cpp.o
[ 10%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/gravity.cpp.o
[ 10%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/zerorate.cpp.o
[ 11%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/heading.cpp.o
[ 11%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/magnetic.cpp.o
[ 11%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/height.cpp.o
[ 12%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/poseupdate.cpp.o
[ 12%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/baro.cpp.o
[ 12%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/gps.cpp.o
[ 13%] Building CXX object hector_localization/hector_pose_estimation_core/CMakeFiles/hector_pose_estimation.dir/src/measurements/rate.cpp.o
[ 13%] Linking CXX shared library /home/yueshukun/hector_quadrotor_tutorial/devel/lib/libhector_pose_estimation.so
[ 13%] Built target hector_pose_estimation
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

gautam-sharma1, 2019-09-26:
I faced the same issue. Try the following.

#!/bin/bash

# to execute when in catkin_ws/src folder
sudo apt-get install ros-kinetic-ros-control
sudo apt-get install ros-kinetic-gazebo-ros-control
sudo apt-get install ros-kinetic-unique-identifier
sudo apt-get install ros-kinetic-geographic-info
sudo apt-get install ros-kinetic-laser-geometry
sudo apt-get install ros-kinetic-tf-conversions
sudo apt-get install ros-kinetic-tf2-geometry-msgs
sudo apt-get install ros-kinetic-joy

git clone -b kinetic-devel https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor
git clone -b catkin https://github.com/tu-darmstadt-ros-pkg/hector_localization
git clone -b kinetic-devel https://github.com/tu-darmstadt-ros-pkg/hector_gazebo
git clone -b kinetic-devel https://github.com/tu-darmstadt-ros-pkg/hector_models
git clone -b catkin https://github.com/tu-darmstadt-ros-pkg/hector_slam

sed -i -e 's/option(USE_PROPULSION_PLUGIN "Use a model of the quadrotor propulsion system" ON)/option(USE_PROPULSION_PLUGIN "Use a model of the quadrotor propulsion system" OFF)/g' hector_quadrotor/hector_quadrotor/hector_quadrotor_gazebo/urdf/CMakeLists.txt

sed -i -e 's/option(USE_AERODYNAMICS_PLUGIN "Use a model of the quadrotor aerodynamics" ON)/option(USE_AERODYNAMICS_PLUGIN "Use a model of the quadrotor aerodynamics" OFF)/g' hector_quadrotor/hector_quadrotor/hector_quadrotor_gazebo/urdf/CMakeLists.txt

# this is to deactivate warnings
sed -i -e 's/add_dependencies(landing_action hector_uav_msgs_generate_message_cpp)//g' hector_quadrotor/hector_quadrotor/hector_quadrotor_actions/CMakeLists.txt

sed -i -e 's/add_dependencies(pose_action hector_uav_msgs_generate_message_cpp)//g' hector_quadrotor/hector_quadrotor/hector_quadrotor_actions/CMakeLists.txt

sed -i -e 's/add_dependencies(takeoff_action hector_uav_msgs_generate_message_cpp)//g' hector_quadrotor/hector_quadrotor/hector_quadrotor_actions/CMakeLists.txt

sed -i -e 's/add_dependencies(hector_quadrotor_controllers hector_uav_msgs_generate_message_cpp)//g' hector_quadrotor/hector_quadrotor/hector_quadrotor_controllers/CMakeLists.txt

cd ..
catkin_make
source devel/setup.bash

It should be fine now

growupboron, 2019-12-24:
f