tu-darmstadt-ros-pkg/hector_quadrotor #81: hector_quadrotor jumping on giving velocity commands

hermesdragon, 2017-09-24:
When I give velocity commands to the quadrotor it jumps. As in, if i only give a velocity command in z direction the goes up with a acceleration and then comes back down. It's almost like firing a projectile. 

All I am doing is : 
Launching the outdoor launch file. 
enabling the motors service
and publishing velocity command to /cmd_vel
Also, even if I publish an all 0 in the /cmd_vel the quadrotor still jumps in the +z-direction a little bit. 

Is this part of the new update too?

hermesdragon, 2017-09-25:
[Here](https://www.youtube.com/watch?v=JEQzEHWO83M) is a video of this problem in hector_quadrotor

davidbcaro, 2018-10-22:
Hi @hermesdragon I'm getting the same error with ROS Kientic, you could fix it?  I appreciate your help.

davidbcaro, 2018-10-22:
> When I give velocity commands to the quadrotor it jumps. As in, if i only give a velocity command in z direction the quadrotor goes up with a acceleration and then comes back down. It's almost like firing a projectile.
> 
> All I am doing is :
> Launching the outdoor launch file.
> enabling the motors service
> and publishing velocity command to /cmd_vel
> Also, even if I publish an all 0 in the /cmd_vel the quadrotor still jumps in the +z-direction a little bit.
> 
> Is this part of the new update too?

Hi @hermesdragon I'm getting the same error with ROS Kientic, you could fix it? I appreciate your help.

hermesdragon, 2018-10-22:
No. I never managed to make it work. There is a problem in the repo I believe. 
Ultimately, I had to go back to Ros indigo and use hector-quadrotor on that.