tu-darmstadt-ros-pkg/hector_quadrotor #85: ROS Hector Quadrotor cmd_vel problem - Quadrotor not responding

sakpancar, 2018-02-20:
[image]

I have installed the packages src folder in the image above. And I compiled packages with "freem100@m100:~/seyit_ws$ catkin_make" freem100@m100:~/seyit_ws$ source devel/setup.bash

I am launching gazebo with "freem100@m100:~/seyit_ws$ roslaunch hector_quadrotor_demo indoor_slam_gazebo.launch" 
[image]

freem100@m100:~/seyit_ws$ rostopic pub /cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.5
angular:
  x: 0.0
  y: 0.0
  z: 0.0"

I share it with a cmd_vel above in a different terminal. But the quadrotor gives no response. Please, what can I do. Please help me...

ppianpak, 2018-03-10:
I think you will need to enable the drone first by calling the service /enable_motors

rohit-s-murthy, 2018-03-12:
When I use `rosservice list` I don't get this service listed. Do I need to do something different?

ppianpak, 2018-03-12:
@rohit-s-murthy That's weird. I don't need to do anything. Have you spawned a drone into your world? Maybe there is a difference in version? I'm using Kinetic by the way.

rohit-s-murthy, 2018-03-12:
@ppianpak Thanks for the reply. Ya I'm using Indigo so maybe that's why it isn't showing. Going through some of the other issues, I guess this service is only in Kinetic.

RyanDirgantara, 2018-05-15:
Did you do the "source devel/setup.bash" on your new terminal?