tu-darmstadt-ros-pkg/hector_quadrotor #96: when can you release the hector_quadrotor for melodic

ShutingKANG, 2018-11-13:
I am really want to ask when could you release the hector_quadrotor for melodic. when I run this file spawn_quadrotor_with_downward_cam.launch, there are always some mistakes. can you fix it? for example there is a error like this :                                                                                                  
unknown macro name: xacro:quadrotor_plugins None None
when processing file: /home/kang/hector_quadrotor/src/hector_quadrotor/hector_quadrotor_description/urdf/quadrotor_downward_cam.gazebo.xacro
RLException: while processing /home/kang/hector_quadrotor/src/hector_quadrotor/hector_quadrotor_gazebo/launch/spawn_quadrotor.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/melodic/lib/xacro/xacro '/home/kang/hector_quadrotor/src/hector_quadrotor/hector_quadrotor_description/urdf/quadrotor_downward_cam.gazebo.xacro' base_link_frame:=/base_link world_frame:=world] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(arg model)' base_link_frame:=$(arg base_link_frame) world_frame:=$(arg world_frame)" name="robot_description"/>
The traceback for the exception was written to the log file

Maybe there is some problem in quadrotor_plugins file.
Thanks for your help.

TONG369, 2019-10-18:
Have you solved your problem? I met exactly the same problem as you? the problem is “unknown macro name: xacro:quadrotor_plugins None None” If so, please contact me, thank you very much~~~

tooHotSpot, 2020-04-14:
Actually there is no limitation to build `hector_quadrotor` for `melodic` from sources. This link to article with general installation steps may be helpful: [Flying with ROS and Hector](https://darienmt.com/autonomous-flight/2018/10/20/flying-ros-and-hector.html).