tu-darmstadt-ros-pkg/hector_quadrotor #108: terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'   what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument Aborted (core dumped)

Jaroan, 2019-12-05:
I have installed Hector Quadrotor outdoor flight demo as described on the [ROS Wiki page](http://wiki.ros.org/hector_quadrotor/Tutorials/Quadrotor%20outdoor%20flight%20demo). Both gazebo and Rviz were working fine for the first time.
[image] 
After that, the above exception was raised and gazebo terminated abruptly without loading.
[image]

Jaroan, 2019-12-05:
This is the terminal output:

[INFO][1575524074.716475, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1575524074.847943973]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1575524074.849594889]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[WARN] [1575524104.737353, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_spawner-5] process has finished cleanly
log file: /home/jasmine/.ros/log/0e91385e-171c-11ea-a434-f828195a1467/controller_spawner-5*.log
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
[gazebo_gui-2] process has died [pid 18251, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/jasmine/.ros/log/0e91385e-171c-11ea-a434-f828195a1467/gazebo_gui-2.log].
log file: /home/jasmine/.ros/log/0e91385e-171c-11ea-a434-f828195a1467/gazebo_gui-2*.log

StefanFabian, 2019-12-05:
Given that the last commit was more than a year ago, I suspect that the plugin may be incompatible with Gazebo 9.
Unfortunately, the crash message is not very specific and would require further debugging.

You could try if it works using kinetic or jade (using Ubuntu 16.04 or Ubuntu 14.04 respectively) using Gazebo 8 or 7.
Alternatively, you would have to build the plugin as Debug and attach a Debugger (I'd recommend CLion or VS Code) to get a stack trace of where exactly it's crashing.
I suspect it's a memory corruption issue based on the exception message.

Jaroan, 2019-12-05:
I guess it's the memory corruption issue. The gazebo had escalated to sigkill. I restarted my system, rerun the commands and it worked.

StefanFabian, 2019-12-05:
Thank you for the update!
Gazebo likes to crash occasionally on startup. Restarting it usually helps.

closed 2019-12-05 by StefanFabian