tu-darmstadt-ros-pkg/hector_quadrotor #109: Contact sensor on Hector Quadrotor

Jean-0101, 2020-04-23:
Hello,

I am trying to add a bump contact sensor to indicate if there is a contact to another object at a time. In other words, I am trying to detect collision. Is there anything implemented on the hector quadrotor or do I have to add a contact sensor? If so, how can I add a contact sensor to the hector quadrotor model? I looked at a lot of tutorials and answers from different sources but could not find anything to fit hector quadrotor demo. I would be happy if someone helps.

Best

Martin-Oehler, 2020-04-24:
You can add a bumber plugin to the robot urdf, see [here](http://gazebosim.org/tutorials?tut=ros_gzplugins#Bumper).

Jean-0101, 2020-04-24:
Hello Martin. Thank you. I will add this bumper plugin to the robot urdf. 

Normally, to add a sensor to a robot model, all we need to do is to add this plugin to the robot's main urdf file? In this case, on Hector Quadrotor Demo, there are plenty of urdf files. Which one is the main urdf file that I need to change? 

quadrotor_sensors.gazebo.xacro on https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/blob/kinetic-devel/hector_quadrotor_gazebo/urdf/quadrotor_sensors.gazebo.xacro

or some other file?

And there is also the case that as far as I understand, we need to add the new plugin to the xacro file and then it is converted to urdf. Do we need any additional files to add like a cpp file for the sensor?

Jean-0101, 2020-04-24:
I changed the quadrotor_sensors.gazebo.xacro file and added the contact_bumper sensor as follows, saved the file, ran the simulation but could not see a new sensor added when I enter rostopic list

<?xml version="1.0"?>

<robot xmlns:xacro="http://ros.org/wiki/xacro">

  
  <gazebo>
    <plugin name="quadrotor_imu_sim" filename="libhector_gazebo_ros_imu.so">
      <updateRate>100.0</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>$(arg base_link_frame)</frameId>
      <topicName>raw_imu</topicName>
      <rpyOffsets>0 0 0</rpyOffsets> 
      <gaussianNoise>0</gaussianNoise>  
      <accelDrift>0.5 0.5 0.5</accelDrift>
      <accelGaussianNoise>0.35 0.35 0.3</accelGaussianNoise>
      <rateDrift>0.1 0.1 0.1</rateDrift>
      <rateGaussianNoise>0.05 0.05 0.015</rateGaussianNoise>
    </plugin>

    <plugin name="quadrotor_baro_sim" filename="libhector_gazebo_ros_baro.so">
      <updateRate>10.0</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>$(arg base_link_frame)</frameId>
      <topicName>pressure_height</topicName>
      <altimeterTopicName>altimeter</altimeterTopicName>
      <offset>0</offset>
      <drift>10.0</drift>
      <gaussianNoise>0.1</gaussianNoise>
    </plugin>

    <plugin name="quadrotor_gazebo_ros_bumper_controller" filename="libgazebo_ros_bumper.so">
      <updateRate>10.0</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>$(arg base_link_frame)</frameId>
      <alwaysOn>true</alwaysOn>    
      <topicName>contact_bumper</topicName>
      <offset>0 0 0</offset>
      <drift>0.0 0.0 0.0</drift>
      <gaussianNoise>1.3e-2 1.3e-2 1.3e-2</gaussianNoise>
    </plugin>

    <plugin name="quadrotor_magnetic_sim" filename="libhector_gazebo_ros_magnetic.so">
      <updateRate>10.0</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>$(arg base_link_frame)</frameId>
      <topicName>magnetic</topicName>
      <offset>0 0 0</offset>
      <drift>0.0 0.0 0.0</drift>
      <gaussianNoise>1.3e-2 1.3e-2 1.3e-2</gaussianNoise>
    </plugin>

    <plugin name="quadrotor_gps_sim" filename="libhector_gazebo_ros_gps.so">
      <updateRate>4.0</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>$(arg base_link_frame)</frameId>
      <topicName>fix</topicName>
      <velocityTopicName>fix_velocity</velocityTopicName>
      <referenceLatitude>49.860246</referenceLatitude>
      <referenceLongitude>8.687077</referenceLongitude>
      <drift>5.0 5.0 5.0</drift>
      <gaussianNoise>0.01 0.01 0.01</gaussianNoise>
      <velocityDrift>0 0 0</velocityDrift>
      <velocityGaussianNoise>0.05 0.05 0.05</velocityGaussianNoise>
    </plugin>

    <plugin name="quadrotor_groundtruth_sim" filename="libgazebo_ros_p3d.so">
      <updateRate>100.0</updateRate>
      <bodyName>base_link</bodyName>
      <topicName>ground_truth/state</topicName>
      <gaussianNoise>0.0</gaussianNoise>
      <frameName>$(arg world_frame)</frameName>
    </plugin>

    <sensor name="gripper_contact_sensor" type="contact">
    <always_on>true</always_on>
    <update_rate>30.0</update_rate>
    <contact>
      <collision>collision</collision>
    </contact>
    <plugin name="gripper_bumper" filename="libgazebo_ros_bumper.so">
      <bumperTopicName>gripper_contact_sensor_state</bumperTopicName>
      <frameName>gripper</frameName>
    </plugin>
  </sensor>

  </gazebo>
</robot>

Jean-0101, 2020-04-24:
After changing the quadrotor_base.urdf.xacro file, I converted it into quadrotor_base.urdf then tried to change into quadrotor_base.sdf file but it says permission denied. Is the quadrotor_base.sdf file protected? How can we add sensors to the quadrotor model?

Martin-Oehler, 2020-04-27:
You do not have to convert the xacro to urdf or sdf, the xacro is converted during execution. You can add the sensor to any xacro, that is loaded. The one you chose should be fine. I can not give you any advice on the bumper plugin, because I have never used it.

If the question is not directly related to hector_quadrotor, you can ask it at [ROS Answers](https://answers.ros.org/questions/).

Jean-0101, 2020-04-27:
So adding the necessary sensor to the quadrotor_base.urdf.xacro file should do the work, I won't need to do anything else. Then I will be able to see the sensor on rostopic list right? 

I asked the question here because I was not sure which xacro file to add the sensor as it is related with the general structure of the project, you say quadrotor_base.urdf.xacro file would be fine. Let me try and tell the result.

Thank you again

Jean-0101, 2020-04-27:
Here is the quadrotor_base.urdf.xacro file. I added the sensor plugin at two places just to be sure

<?xml version="1.0"?>

<robot xmlns:xacro="http://ros.org/wiki/xacro">

  <xacro:include filename="$(find hector_sensors_description)/urdf/sonar_sensor.urdf.xacro" />
  <xacro:property name="pi" value="3.1415926535897931" />

  
  <xacro:macro name="quadrotor_base_macro">
    <link name="base_link">
      <inertial>
        <mass value="1.477" />
        <origin xyz="0 0 0" />
        <inertia ixx="0.01152" ixy="0.0" ixz="0.0" iyy="0.01152" iyz="0.0" izz="0.0218" />
      </inertial>

      <visual>
        <origin xyz="0 0 0" rpy="0 0 0" />
        <geometry>
          <mesh filename="package://hector_quadrotor_description/meshes/quadrotor/quadrotor_base.dae"/>
        </geometry>
      </visual>

      <collision>
        <origin xyz="0 0 0" rpy="0 0 0" />
        <geometry>
          <mesh filename="package://hector_quadrotor_description/meshes/quadrotor/quadrotor_base.stl"/>
        </geometry>
      </collision>

    <gazebo>
     <plugin name="${name}_gazebo_ros_bumper_controller" filename="libgazebo_ros_bumper.so">
       <alwaysOn>true</alwaysOn>
       <updateRate>${update_rate}</updateRate>
       <bumperTopicName>${name}_bumper</bumperTopicName>
       <frameName>world</frameName>
     </plugin>
     </gazebo>

    </link>

    
    <xacro:sonar_sensor name="sonar" parent="base_link" ros_topic="sonar_height" update_rate="10" min_range="0.03" max_range="3.0" field_of_view="${40*pi/180}" ray_count="3">
      <origin xyz="-0.16 0.0 -0.012" rpy="0 ${90*pi/180} 0"/>
    </xacro:sonar_sensor>
    
    <gazebo>
     <plugin name="${name}_gazebo_ros_bumper_controller" filename="libgazebo_ros_bumper.so">
       <alwaysOn>true</alwaysOn>
       <updateRate>${update_rate}</updateRate>
       <bumperTopicName>${name}_bumper</bumperTopicName>
       <frameName>world</frameName>
     </plugin>
    </gazebo>

  </xacro:macro>
</robot>

After that I saved the project and run with roslaunch hector_quadrotor_demo indoor_slam_gazebo.launch command. There was no difference with the sensor list. I closed the project and typed

catkin_make
source devel/setup.bash

Then I re run the project. The final list of the sensors are as follows 

user:~$ rostopic list
/aerodynamics/wrench
/altimeter
/clock
/cmd_vel
/command/motor
/command/twist
/command/wrench
/fix
/fix_velocity
/front_cam/camera/camera_info
/front_cam/camera/image
/front_cam/camera/image/compressed
/front_cam/camera/image/compressed/parameter_descriptions
/front_cam/camera/image/compressed/parameter_updates
/front_cam/camera/image/compressedDepth
/front_cam/camera/image/compressedDepth/parameter_descriptions
/front_cam/camera/image/compressedDepth/parameter_updates
/front_cam/camera/image/theora
/front_cam/camera/image/theora/parameter_descriptions
/front_cam/camera/image/theora/parameter_updates
/front_cam/parameter_descriptions
/front_cam/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/ground_truth/state
/ground_truth_to_tf/euler
/ground_truth_to_tf/pose
/joint_states
/magnetic
/motor_pwm
/motor_status
/pressure_height
/propulsion/wrench
/quadro/trigger
/raw_imu
/rosout
/rosout_agg
/scan
/sonar_height
/supply
/tf
/tf_static
/wind

Again the contact sensor was not there. Can it be because I am working on the constructsim environment or something else?

Martin-Oehler, 2020-04-27:
Your issue is probably related to your usage of the bumper plugin. You might want to look into, if you are using it correctly. However, I can not help you any further in this regard.

Jean-0101, 2020-04-27:
Alright Martin. Vielen dank! :)