tu-darmstadt-ros-pkg/hector_quadrotor #110: Turning on motors in hector quadrotor demo

Jean-0101, 2020-04-25:
hello everyone,

I have run hector quadrotor indoor demo and able to move the robot in all directions. When I look at the motor status, which needs to show voltage and current on the motors, all data is displayed as 0, on and running parameters are set to False.

How can I enable the motors such that they show voltage frequency current values when I type rostopic echo /motor_status ?

I tried to enable the motors with rosservice call /enable_motors "enable: true" written as in tutorial but it gives me the error:

user:~$ rosservice call /enable_motors "enable: true" ERROR: Service [/enable_motors] is not available.

The /motor_status data is as follows:

header: 
seq: 36144 
stamp: secs: 361 
nsecs: 450000000 frame_id: '' 
on: False 
running: False 
voltage: [0.0, 0.0, 0.0, 0.0] 
frequency: [0.0, 0.0, 0.0, 0.0]
current: [0.0, 0.0, 0.0, 0.0]

^Cheader: seq: 36145 stamp: secs: 361 nsecs: 460000000 frame_id: '' on: False running: False voltage: [0.0, 0.0, 0.0, 0.0] frequency: [0.0, 0.0, 0.0, 0.0] current: [0.0, 0.0, 0.0, 0.0]

Martin-Oehler, 2020-04-27:
We do not use this stack anymore, therefore, we can only give basic support advice. Have you checked, if the service is available in a different namespace?

Jean-0101, 2020-04-27:
Thank you very much for your help. This is the list of services available currently. Did you mean the enable motor service can be available as one of these?

rosservice list
/controller_manager/list_controller_types
/controller_manager/list_controllers
/controller_manager/load_controller
/controller_manager/reload_controller_libraries
/controller_manager/switch_controller
/controller_manager/unload_controller
/controller_spawner/get_loggers
/controller_spawner/set_logger_level
/engage
/front_cam/camera/image/compressed/set_parameters
/front_cam/camera/image/compressedDepth/set_parameters
/front_cam/camera/image/theora/set_parameters
/front_cam/set_camera_info
/front_cam/set_parameters
/gazebo/apply_body_wrench
/gazebo/apply_joint_effort
/gazebo/clear_body_wrenches
/gazebo/clear_joint_forces
/gazebo/delete_light
/gazebo/delete_model
/gazebo/get_joint_properties
/gazebo/get_light_properties
/gazebo/get_link_properties
/gazebo/get_link_state
/gazebo/get_loggers
/gazebo/get_model_properties
/gazebo/get_model_state
/gazebo/get_physics_properties
/gazebo/get_world_properties
/gazebo/pause_physics
/gazebo/reset_simulation
/gazebo/reset_world
/gazebo/set_joint_properties
/gazebo/set_light_properties
/gazebo/set_link_properties
/gazebo/set_link_state
/gazebo/set_logger_level
/gazebo/set_model_configuration
/gazebo/set_model_state
/gazebo/set_parameters
/gazebo/set_physics_properties
/gazebo/spawn_sdf_model
/gazebo/spawn_urdf_model
/gazebo/unpause_physics
/ground_truth_to_tf/get_loggers
/ground_truth_to_tf/set_logger_level
/raw_imu/calibrate
/raw_imu/set_accel_bias
/raw_imu/set_rate_bias
/robot_state_publisher/get_loggers
/robot_state_publisher/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/shutdown

Jean-0101, 2020-04-27:
There is also a similar issue 

https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor/issues/66

Martin-Oehler, 2020-04-27:
It does not look like the service is available, for some reason. Unfortunately, I can't help you with this problem. You should probably take a look at the source code and debug the issue.

Jean-0101, 2020-04-27:
Maybe it is related with I am working on theconstructsim environment. I will try to run at my local and see if there is any change. Alright Martin thank you anyway.

Jean-0101, 2020-04-27:
Can it be related with the version of the system and gazebo? I am working on Ubuntu 16.04 Gazebo 7. Maybe it can run better in other versions. What was the version you successfully ran the /enable_motors service?

Jean-0101, 2020-04-27:
Release date of Ubuntu 16.04 was on 2016 , your project page has been updated on 2014, you probably have developed the hector quadrotor demo on a previous version :) Would it make a difference?

Martin-Oehler, 2020-04-27:
> Can it be related with the version of the system and gazebo? I am working on Ubuntu 16.04 Gazebo 7.

As long as everything builds fine and does not throw errors on start-up, it should probably run. You could test Ubuntu 14.04. 

> What was the version you successfully ran the /enable_motors service?

I have never used this package.

Jean-0101, 2020-04-27:
Ok thank you for your help on this old issue. Do you know who has implemented these services? Is there anyone that I can ask?